ceoloide/ergogen-footprints #1: Jumpers dont rotate with the controller

Dollehz, 2024-02-20:
When the niceNano controller is at an angle the jumper pads stay oriented horizontal.

[image]

ceoloide, 2024-02-20:
Thanks for reporting! I believe the rotation parameter.is missing, will modify ASAP.

I am creating a test / demo board for the footprints to uncover problems like these at https://github.com/ceoloide/ergogen-footprints-demo and will soon link it to this repo. If you have any advice on what to test for, let me know. I plan to fabricate the board to validate footprints in real life.

ceoloide, 2024-02-21:
I've committed the changes necessary to enable dynamic rotation for all elements of the footprint, since silkscreen and labels were also not rotating.

To update the submisule and get the latest code, I use this command:

```bash
git submodule update --init --remote
```

Will track this for other footprints in an another issue. Thank you so much for reporting it!

Here's a preview of the change:
[image]

closed 2024-02-21 by ceoloide