ceoloide/ergogen-footprints #3: Feature: Add extra pins to footprint for nice!nano Request

TildeWill, 2024-02-24:
I'd like to use the three middle pins and the battery terminals on my board. 
Specifically pins 101, 102, 107 and B+ and B- shown here: https://nicekeyboards.com/docs/nice-nano/pinout-schematic
I'm using the nice!nano as a reversible footprint for my split design. 

Note: the battery terminals are smaller holes, not full pins, for direct soldering of wires. I'm not sure what that means for connecting them to a PCB so a battery connector can be used elsewhere on the PCB; maybe I should just solder my battery directly to it and forget about it?

ceoloide, 2024-04-04:
I've added the option to use the nice!nano footprint on a single side, where there is enough space in between sockets to add the three extra pins (1.01, 1.02, 1.07).

With the reversible footprint, the position of the extra pins would interfere with the solder jumpers, and would require even more space to have jumpers for those extra pins. For now I don't plan to add this to the footprint, since it's a niche use case.

Re. the battery terminals: you can already wire a battery connector to RAW and GND, and it would charge the battery. Those sockets are dedicated to direct wiring and are not useful for socketing (they are also smaller than regular socket pins).

ceoloide, 2024-04-07:
I've figured that the extra pins are easy to add in the reversible mode with only minimum jumpers (i.e. first 4 rows). To be added to the mode with all pins having jumpers, I will need to move around some traces, but I figured it's completely doable, so I am reopening this issue.

bptrent, 2024-11-22:
Note the B+ and B- are the same as  RAW and GND so having them int the footprint isn't required for the battery functionality. Having the 3 low frequency pins that have been added should be enough for this issue to close.

ceoloide, 2024-11-22:
Pins are all added, I just need to make the following changes:

1) ensure pin labels are present for all extra pins and for the 10th row

2) try to have reversible routes even when extra pins are present (nice to have)