ceoloide/ergogen-footprints #9: Add 3D models to footprint definition

ceoloide, 2024-03-28:
Add optional 3D models to the footprint (perhaps reusing Scotto's ones)

Grazfather, 2024-03-31:
#10 and #11 added these for Choc v1/v2

hyperpuncher, 2024-05-07:
#29 added model options for power switch, nice!nano and keycaps to all switches

peterjc, 2025-09-01:
Would adding the parameters:

```
    keycap_3dmodel_filename: '',
    keycap_3dmodel_xyz_offset: [0, 0, 0],
    keycap_3dmodel_xyz_rotation: [0, 0, 0],
    keycap_3dmodel_xyz_scale: [1, 1, 1],
```

to all the remaining footprints be reasonable, even if there isn't a 3D model yet?

e.g. I might try using the Nice!Nano model as a substitute for visualizing the Supermini NRF52840.

peterjc, 2025-09-05:
It looks like the original https://github.com/infused-kim/kb_ergogen_fp/blob/main/conn_molex_pico_ezmate_1x02.js has been updated with 3D models, so the forked version here could use those too: https://github.com/ceoloide/ergogen-footprints/blob/main/battery_connector_molex_pico_ezmate_1x02.js