ceoloide/ergogen-footprints #18: Upgrade footprints to KiCad 8

ceoloide, 2024-04-10:
KiCad 8 will become the minimum version supported by this footprint library, now that [Ergogen v4.1.0](https://github.com/ergogen/ergogen/tree/v4.1.0) has been released with support for KiCad 8 PCB template injection.

simisimis, 2024-04-24:
thank you for your amazing work!

I was trying to use your updated library with kicad 8.0.1
When using `switch_choc_v1_v2.js` with default values, ergogen generates pcb file that wouldn't open. It throws an error:
```
Error loading PCB './proj.kicad_pcb'. Expecting 'end' in './proj.kicad_pcb', line 526, offset 32.
```
while the line in that file is:
(fp_arc (start 2.52 -6.2) (mid 2.139878 -6.382304) (end 2.0 -6.78) (layer "B.SilkS") (stroke (width 0.15) (type solid)))

I tried figuring it out myself if some bracket was missing, but had to give up.
If I revert the [changes(7e4fc81)](https://github.com/ceoloide/ergogen-footprints/commit/7e4fc81f92799f52cd9a8f7de64685dca26df2c6) you made for that footprint, then kicad is able to open ergogen generated file.

ceoloide, 2024-04-25:
Hello @simisimis!

The error you quote is shown by KiCad when you open a KiCad 5 `*.kicad_pcb` file containing KiCad 8 footprints that have incompatible syntax (like `fp_arc` which changed syntax from V5 to V8).

I updated the `README.md` to explicitly call out how to force a KiCad 8 file:

- Make sure you have ergogen v4.1.0 or greater
- Have `template: kicad8` in your pcb definition (see `README.md`
- Use KiCad 8 or greater to open the file

I won't be supporting KiCad 5 moving forward, but should you need KiCad 5 compatible footprints, there is a version and branch tagged specifically with that.

closed 2024-04-25 by ceoloide

simisimis, 2024-04-25:
thank you for the explanation. I was using ergogen 4.1 to generate pcb files and trying to open them with kicad 8.  What I was missing is to put template: kicad8.
Just tried adding it following your updated readme and it worked! :pray: