ceoloide/ergogen-footprints #32: display_nice_view has nets swapped when reversible is false

Grazfather, 2024-06-10:
```yaml
     oled:
         what: ceoloide/display_nice_view
         params:
           VCC: VCC
           MOSI: OLED_MOSI
           CS: OLED_SS
           SCK: OLED_SCK
           include_silkscreen: true
```
[image]

hackrmomo, 2024-08-11:
The silkscreen legend is correct if I'm not mistaken according to this: 
[image]

You should be able to temporarily circumvent this by doing 
```yaml
params:
  MOSI: CS
  SCK: GND
  VCC: VCC # no actual change needed
  GND: SCK
  CS: MOSI
```

Grazfather, 2024-09-29:
Yep, unfortunately I didn't notice until after I had my pcb made! But I was able to use it by rotating the display.

ceoloide, 2024-10-21:
Sorry, I haven't yet tested single-side footprints that much, and I missed these kind of things. Will slowly test everything and fix it!

commit cb995ca7 (2024-11-30): Overhaul nets and labels for Nice!View and SSD1306. Closes #32
closed 2024-11-30 by ceoloide