ceoloide/ergogen-footprints #34: Missing silkscreen text on mcu nice nano

ujl123, 2024-08-08:
Missing silk screen at the bottom left of the nice nano.  
[image]

Is this intentional to help users key/align the mcu? Or are those pins not to be used or just mistakenly left off?

ceoloide, 2024-10-15:
Sorry for the long delay, I was on paternity leave and still working to claw back some free time from my 9 mo baby.

The lack of label is not intentional, and it's a bug. This is a result of trying to have the ability to add or leave out the additional three pins the Nice!Nano has.

ceoloide, 2024-11-11:
This should now be resolved. both on the missing silkscreen, and the silkscreen appearing on both sides when a single side MCU option is chosen.

To override the single-side silkscreen behavior, there is the `show_silk_labels_on_both_sides` parameter (default `false`).

closed 2024-11-11 by ceoloide