ceoloide/ergogen-footprints #36: TRRS footprint "side" parameter doesn't do anything

monlovesmango, 2024-10-09:
hey there,

I was working with the trrs footprint and am wanting to mount it on the back. I tried using the `side` parameter but it doesn't seem to actually do anything. 

since the code was already super organized and modularized it was really easy for me to implement the `side` parameter myself. let me know if you'd like a pr for this update. this is my first foray into ergogen and makerjs so would definitely need some review from you.

thanks for all the great footprints and resources! its been immensely helpful to me :)

ceoloide, 2024-10-15:
Can you paste here the parameters you have used in your config and a screenshot of the PCB? It may be bacause the footprint is through-hole and with symmetric option it just looks the same on both sides.

Be warned I have not tested single-side options as much.

ceoloide, 2024-10-21:
This is now fixed, thank you for reporting the issue!

[image]

closed 2024-10-21 by ceoloide

monlovesmango, 2024-10-22:
thank you for your great footprint library!