ceoloide/ergogen-footprints #49: Add Kailh PG1316S and PG1316M

sommerper, 2025-02-25:
Hi!

Without knowing how much work goes into this and if it's even possible - would it take a lot of work to add Kailh PG1316S and PG1316M to the footprints?

The only files I found are for Kicad:
https://github.com/mikeholscher/zmk-config-mikefive/tree/main/files/footprint-and-cad

But I don't know if this is something you can use...

Thanks!

jusdisgi, 2025-04-17:
It's not of a quality befitting the official collection, but I made a footprint for the PG1316S. It's available here: https://github.com/jusdisgi/splave-ferris/blob/main/splave-ferris_pg1316s/footprints/switch_pg1316s.js

If you have the kicad file for the M version I would like to have it. And I'll make an ergogen footprint of it for your trouble.

Also, I didn't use it for this one, but this converter may come in handy: https://kicad2ergogen.genteure.com/
...one weird note: there's some kind of bug and you need to remove one line from the reslulting file...it's an f.push line that says "version" with a date. No idea why that breaks it, but apparently it's a known issue.

sommerper, 2025-04-22:
@jusdisgi Thanks a bunch! I'll see if I can update one of my ergogen configs for this.

I'll keep an eye out for the M version but there doesn't seem to be much floating around for the time being...