ceoloide/ergogen-footprints #54: Are KiCad sources available somewhere?

AnthonyVH, 2025-03-17:
The README explains that all the ergogen code is based on KiCad footprints/symbols. It seems that these are not bundled though. Would it be possible to make these available (or is there a straightforward way to extract them from the Ergogen files)? I'm particularly interested in the Supermini NRF52840 and the Gateron KS33 hotswap sockets.

jusdisgi, 2025-06-07:
Those are both available in the marbastlib library, see here: https://github.com/ebastler/marbastlib/tree/main

that one is invaluable for anybody doing keyboards with kicad.

peterjc, 2025-06-17:
How about https://github.com/ceoloide/ergogen-footprints/blob/main/switch_choc_v1_v2.js and https://github.com/ceoloide/ergogen-footprints/blob/main/mounting_hole_plated.js please?

I'm using those with Ergogen v4.1.0 (thank you!), and have laid out my traces in KiCad v9.0.2 on macOS, then:

1. Menu "File", "Fabrication Outputs...", "Gerbers (.gbr)"
2. Click "Run DCR..." (button in bottom left)
3. Lots of warnings: `The current configuration does not include the footprint library 'ceoloide'.` (one warning per footprint used)

I guess I need to download KiCad versions of those, and tell KiCad about them here:

1. Menu "Preferences", "Manage Footprint Libraries..."
2. Add folder of `.kicad_mod` files (probably under the "Project Specific Libraries" tab)

jusdisgi, 2025-06-17:
You can safely ignore those warnings.

peterjc, 2025-06-17:
OK, thanks @jusdisgi